intel/zephyr.js #1594: [board] board is not supported in qemu

qiaojingx, 2017-10-16:
### Description

Building samples and tests of Board API failed in qemu. Other tests which used board api are also failed.
 
### Test Code

[test-board.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board.js);
[test-board-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board-manual.js);
[Board.js](https://github.com/01org/zephyr.js/blob/master/samples/Board.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js);
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js);

### Steps to Reproduction

1. `make BOARD=qemu_x86 qemu JS=samples/Board.js`
 
### Actual Result

[image]
 
### Expected Result

 
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fae50ff - 1d5a99d | qemu | Oct 16, 2017 | Fail
master | 6eedbe4 | qemu | Oct 16, 2017 | Pass
### Additional Information

grgustaf, 2017-12-01:
Works for me on master, can you try again @qiaojingx?

qiaojingx, 2017-12-04:
Verified with commit cbec79d in master. This issue is fixed. Close.

closed 2017-12-04 by qiaojingx