intel/zephyr.js #1621: [ashell] ble cannot be enabled again after execute 'stop'

qiaojingx, 2017-11-01:
### Description

After stopping samples/BLE.js in ashell, bluetooth cannot be re-enabled.
 
### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
[BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js) 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3.  `$ screen /dev/ttyACM0 115200` to connect to the ashell from a terminal
4. `acm> load test.js` and Enter
5. `RAW> /*Paste code of BLE.js */ `
6. `acm> run test.js`
7. `acm> stop`
8. `acm> load test.js`
### Actual Result

Failed to enable bluetooth.
[image]

 If connect to this ble server with phone, error message will print.
[image]

### Expected Result

BLE can be enabled without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 6d751ea | Arduino 101 | Nov 1, 2017 | Fail
 
### Additional Information

This issue is also reproduced in Zephyr IDE.

jimmy-huang, 2017-11-02:
this is a know issue, the reason is that Zephyr doesn't have a way to disable BT after it has been up, thus we added a error message when you run this in the IDE that you'll need to reboot the board in order to get the BLE sample to work again.  So this will be won't fix until Zephyr has added support to disable/re-enable BT.

qiaojingx, 2017-11-09:
Closed since Zephyr doesn't support to disable/re-enable BT

closed 2017-11-09 by qiaojingx