intel/zephyr.js #1627: [spi_screen] Out of memory when draws a solid rectangle

cuiyanx, 2017-11-02:
### Description

When test [/samples/SPI_Screen.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI_Screen.js), throw out an error.

Error log:
```
511:(ERROR) malloc(40960) failed
SPI error: argument cannot have an error flag
```

 

### Test Code

[/samples/SPI_Screen.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI_Screen.js)

 

### Steps to Reproduction

 

### Actual Result

Throw out an error.
 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e19ed4f | Arduino 101 | Nov 1, 2017 | Fail

 

### Additional Information

grgustaf, 2017-12-01:
@brianjjones is this fixed by your new frameless mode for A101?

grgustaf, 2017-12-01:
@cuiyanx was this on A101 I assume? That matters because of the low available RAM.

cuiyanx, 2017-12-04:
@grgustaf Yes, I think so.
@brianjjones Can we work around this?

cuiyanx, 2017-12-19:
Fixed and closed.

closed 2017-12-19 by cuiyanx