intel/zephyr.js #1629: [gfx] Missing GFX module

cuiyanx, 2017-11-03:
### Description

When test [/samples/SPI_Screen.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI_Screen.js), throw out an error.

Error log:
`NotSupportedError: module not found: 'gfx'`

 

### Test Code

 [/samples/SPI_Screen.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI_Screen.js)

 

### Steps to Reproduction

 

### Actual Result

Throw out error.
 

### Expected Result

No error.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82c3083c9 | Arduino 101 | Nov 3, 2017 | Fail

 

### Additional Information

brianjjones, 2017-11-03:
This is because we switched to using Jerry modules and GFX wasn't set up for that yet.  #1636 will take care of this issue.

cuiyanx, 2017-11-07:
Fixed and closed

closed 2017-11-07 by cuiyanx