intel/zephyr.js #1631: [Mac] Fail to build qemu target on Mac

qiaojingx, 2017-11-03:
### Description

Building qemu target on Mac failed. 

### Test Code

[HelloWorld.js](https://github.com/01org/zephyr.js/blob/master/samples/HelloWorld.js)

### Steps to Reproduction

1. Building qemu on MAC by follow instructions: [building-qemu-and-arduino-101-targets](https://github.com/01org/zephyr.js#building-qemu-and-arduino-101-targets)

`make JS=samples/HelloWorld.js BOARD=qemu_x86 qemu`

### Actual Result

Linker error:
[image]

 ### Expected Result

Building successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82c3083 | qemu | Nov 3, 2017 | Fail
master | 4b77505 | qemu | Oct 11, 2017 | Pass

### Additional Information

jimmy-huang, 2017-11-04:
@qiaojingx With Zephyr 1.9, the mac instructions has been slightly updated, can you see if you can build qemu with these instructions?  I had to re-build crosstool-ng from scratch, but now I cannot even finish building crosstool-ng.

qiaojingx, 2017-11-07:
I also run into trouble when executing `ct-ng build` by following the latest instructions on setting up cross tool-ng with 1.9.

jimmy-huang, 2017-11-07:
ok, is this the same issue you are seeing? There's a work-around here

https://github.com/zephyrproject-rtos/zephyr/issues/3683

qiaojingx, 2017-11-08:
Thanks, I can finish `ct-ng build` now. But there is only one folder named `i586-zephyr-elf` under x-tools, while make qemu target need `i586-pc-elf`not `i586-zephyr-elf`. So make qemu target still failed. How can I fix it?

jimmy-huang, 2017-11-08:
what you need to do is change in the deps/zephyr/scripts/Makefile.toolchain.xtools
from CROSS_COMPILE_TARGET_x86 = i586-pc-elf
to CROSS_COMPILE_TARGET_x86 = i586-zephyr-elf

qiaojingx, 2017-11-09:
After modify the `deps/zephyr/scripts/Makefile.toolchain.xtools` as above, another error will occurs. 
`/Users/zjs/zephyr.js/deps/jerryscript/jerry-core/jrt/jrt.h:19:19: fatal error: stdio.h: No such file or directory`
 
But if make another change in `deps/zephyr/scripts/Makefile.toolchain.xtools` as below:
From `TOOLCHAIN_CFLAGS = -I${CROSS_COMPILE_ROOT}/${CROSS_COMPILE_TARGET}/sysroot/usr/include` 
to `TOOLCHAIN_CFLAGS = -I${CROSS_COMPILE_ROOT}/${CROSS_COMPILE_TARGET}/include`.  
make qemu target will succeed.

qiaojingx, 2017-11-09:
Or modify `.config` file as below as well before `ct-ng build`:
```
# General toolchain options
#
CT_FORCE_SYSROOT=y
CT_USE_SYSROOT=y
CT_SYSROOT_NAME="sysroot"
CT_SYSROOT_DIR_PREFIX=""
#CT_WANTS_STATIC_LINK=y
CT_TOOLCHAIN_PKGVERSION=""
CT_TOOLCHAIN_BUGURL=""

#
# Tuple completion and aliasing
#
CT_TARGET_VENDOR="pc"
CT_TARGET_ALIAS_SED_EXPR=""
CT_TARGET_ALIAS=""
```
Then, no need to change zephyr.js. make qemu target will succeed too.

jimmy-huang, 2017-11-10:
This is good to know, I'll test it out, but I also hit an error buliding cross debugger, I had to disable it in the .config, did you run into that issue? or this patch fixes it as well?

qiaojingx, 2017-11-10:
I met 4 issues totally. (met 1-3 when using zephyr 1.9, met 4 when using zephyr latest master).
1. `stat: illegal option -- c` workaround: zephyrproject-rtos/zephyr#3683
2.  `Build failed in step 'Extracting and patching toolchain components'` workaround: replace `binutils-2.28.tar.xz` under `/CrossToolNG/build/.build/tarballs/` with right one.
3. `configure.ac:25: error: version mismatch.  This is Automake 1.15.1` workaround: edit `/Volumes/CrossToolNG/build/.build/src/gettext-0.19.8.1/gettext-tools/src/configure.ac`
4. `Unknown stat format options` workaround: `ct-ng oldconfig` before edit `.config`

Except these issues, I didn't hit others.

jimmy-huang, 2017-11-10:
Weird, I didn't meet 2, but I have met a couple more, cross_gdb didn't build which I have to disable, and now I am seeing this when I build qemu_x86
/Users/zhuang1/projects/zephyrjs/zephyr.js/deps/jerryscript/jerry-core/jrt/jrt.h:19:19: fatal error: stdio.h: No such file or directory
 #include <stdio.h>

jimmy-huang, 2017-11-10:
@qiaojingx ok, I've got it to build after removing everything and restart.  But I am still hitting the gdb issue, so i have to comment out CT_GDB_CROSS.  Btw, I created a patch #1651 to update the build instructions, can you review it?

qiaojingx, 2017-11-13:
I have reviewed #1651, it looks good to me.

PR #1651 (2017-11-13): [build] Updated MacOS build instructions
Updated instructions with work-arounds when building on MacOS
with Zephyr 1.9.

Fixes #1631
commit f54075cd (2017-11-13): [build] Updated MacOS build instructions (#1651)

Updated instructions with work-arounds when building on MacOS
with Zephyr 1.9.

Fixes #1631
closed 2017-11-13 by grgustaf