intel/zephyr.js #1637: [webusb] webusb module is missing

qiaojingx, 2017-11-06:
### Description

This is a regression issue. After patch #1575. webusb module is missing.
 
### Test Code

[WebUSB.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSB.js);
[WebUSBConfig.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSBConfig.js); 
[WebUSBConfigGroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSBConfigGroveLCD.js); 
[test-webusb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-webusb-manual.js); 

### Steps to Reproduction

1. `make JS=samples/WebUSB.js`
2. `make dfu`
### Actual Result

[image]
 
### Expected Result

Sample runs successfully and  a pop up notification displays in chrome.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a8ff356 - eaee8d5 | Arduino 101 | Nov 6, 2017 | Fail
master | 6149191 | Arduino 101 | Nov 6, 2017 | Pass
### Additional Information

brianjjones, 2017-11-07:
This should be fixed by #1642

qiaojingx, 2017-11-08:
Verified with commit 687af46 + #1642. This issue can be fixed.
I will close it after patch merged.

qiaojingx, 2017-11-15:
Verified with commit 8fe2e17. This issue is fixed. Close

closed 2017-11-15 by qiaojingx