intel/zephyr.js #1638: [event] Type Error occurs in Event

qiaojingx, 2017-11-06:
### Description

This is a regression issue. After patch #1575. `Error: handle was incorrect type` occurs when running samples/Events.js and tests/test-event.js
 
### Test Code

[Events.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/Events.js);
[test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js); 

### Steps to Reproduction

1. `make JS=samples/tests/Events.js`
2. `make dfu`
### Actual Result

A101 and K64F:
[image]

Linux:
[image]

### Expected Result

Sample runs successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a8ff356 - eaee8d5 | Arduino 101 & FRDM-K64F & Linux | Nov 6, 2017 | Fail
master | 6149191 | Arduino 101 & FRDM-K64F & Linux | Nov 6, 2017 | Pass
### Additional Information

brianjjones, 2017-11-07:
This should be fixed by #1642

qiaojingx, 2017-11-08:
Verified with commit 687af46 + #1642. TypeError in Events can be fixed.
But TypeError in BLE is still there.

brianjjones, 2017-11-15:
The updated version of #1642 should take care of the issue for BLE as well.

qiaojingx, 2017-11-15:
Verified with commit 8fe2e17. This issue is fixed. Close

closed 2017-11-15 by qiaojingx