intel/zephyr.js #1639: [dhcp] Failed to build with net-config module.

qiaojingx, 2017-11-06:
### Description

This is a regression issue. After patch #1575 merged, unable to build samples and tests which required `net-config` module.
 
### Test Code

[TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js);
[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js); 
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js); 
[test-tcp4-server-DHCP.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server-DHCP.js); 
[test-ws4-server-dhcp.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws4-server-dhcp.js);
 
### Steps to Reproduction

1. `make JS=samples/TCPEchoServ4_DHCP.js`
2. `make BOARD=frdm_k64f JS=samples/TCPEchoServ4_DHCP.js`

### Actual Result

Unable to build:
[image]
[image]

### Expected Result

Build succeed without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a8ff356 - eaee8d5 | Arduino 101 & FRDM-K64F | Nov 6, 2017 | Fail
master | 6149191 | Arduino 101 & FRDM-K64F | Nov 6, 2017 | Pass
### Additional Information

brianjjones, 2017-11-07:
It looks like '-' isn't a valid character for Jerry modules, I've changed it to '_' to fix this.

brianjjones, 2017-11-07:
This will be fixed by #1643

qiaojingx, 2017-11-08:
Verified with commit 687af46 + #1643 . This issue can be fixed.
I will close it after patch merged.

qiaojingx, 2017-11-17:
Verified with commit b81ee15. This issue is fixed. Close

closed 2017-11-17 by qiaojingx