intel/zephyr.js #1640: [callback] Callbacks function of on event isn't triggered.

qiaojingx, 2017-11-06:
### Description

This is a regression issue. After patch #1575 merged. Callbacks function of on event isn't triggered. It effects websocket, tcp and ocf.
 
### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[test-ws4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws4-server.js); 
[test-ws6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws6-server.js); 
[test-tcp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server.js); 
[test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js); 
[test-tcp4-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-client.js)
### Steps to Reproduction

Take WebSocketServer.js on K64F as an example:
1. `make BOARD=frdm_k64f JS=samples/websockets/WebSocketServer.js`
2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`
3. add router and address on Linux
```
ip -6 route add 2001:db8::/64 dev eno1
ip addr add 2001:db8::2 dev eno1
```
4. `node samples/websockets/NodeWebSocketClient.js`

### Actual Result

WebSocketServer.js: 
[image]

test-ws4-server.js & test-ws6-server.js:
[image]

test-tcp4-server.js & test-tcp6-server.js：
[image]

### Expected Result

WebSocketServer.js: 
[image]

test-ws4-server.js & test-ws6-server.js:
[image]

test-tcp4-server.js & test-tcp6-server.js：
[image]
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a8ff356 - eaee8d5 | Arduino 101 & FRDM-K64F | Nov 6, 2017 | Fail
master | 6149191 | Arduino 101 & FRDM-K64F | Nov 6, 2017 | Pass
### Additional Information

brianjjones, 2017-11-08:
I don't have a router to test this, but it might be fixed by #1642

qiaojingx, 2017-11-08:
Verified with commit 687af46 + #1642. This issue can be fixed.
I will close it after patch merged.

brianjjones, 2017-11-15:
@qiaojingx  thanks, #1642 has now been merged.

qiaojingx, 2017-11-15:
Verified with commit 8fe2e17. This issue is fixed. Close

closed 2017-11-15 by qiaojingx