intel/zephyr.js #1645: [timers] Output of Timers is not in order on Linux

qiaojingx, 2017-11-08:
### Description

This is a regression issue. The output order of Timers.js changes each time on Linux.

### Test Code

[Timers.js](https://github.com/01org/zephyr.js/blob/master/samples/Timers.js)
 
### Steps to Reproduction

1.  `make BOARD=linux`
2.  `./outdir/linux/release/jslinux samples/Timers.js`

### Actual Result

[image]
 
[image]

### Expected Result

[image]

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 687af46 | Linux | Nov 8, 2017 | Fail
master | 6d751ea | Linux | Nov 8, 2017 | Pass
 
### Additional Information

jimmy-huang, 2017-11-08:
This regression is caused by reverting the linux timers back to old way of doing timers, since the new timers used in Zephyr is more accurate but it doesn't build for linux on Mac.  So until we find a way to add async timers support on Mac, the linux timers will be inaccurate.

grgustaf, 2017-12-01:
Marking as low priority because Linux is just a testing target, not a real use case.