intel/zephyr.js #1648: [gfx] If 'y1 > 100' on drawLine(x0, y0, x1, y1, color), the positioning error is too large

cuiyanx, 2017-11-09:
### Description

If 'y1 > 100' on drawLine(x0, y0, x1, y1, color), the y1 positioning error is too large.
 

### Test Code

Line 31~33 on [samples/SPI_Screen.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI_Screen.js)

Simple test:
```
var LCD = require("ST7735.js");
var gfxLib = require("gfx");

var GFX = gfxLib.init(LCD.width, LCD.height, LCD.initScreen, LCD.drawCB, LCD);

GFX.drawLine(48, 50, 96, 160, [0x00, 0x00]);
console.log("\ndrawLine(x0, y0, x1, y1, color): ");
console.log("expected result: an oblique line");
```

 

### Steps to Reproduction

 

### Actual Result

Unable to reach the y1 point.
 

### Expected Result

Reach the y1 point.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 687af4600 | Arduino 101 | Nov 9, 2017 | Fail

 

### Additional Information

brianjjones, 2017-12-29:
Seems to be working for me on the latest, @cuiyanx can you retest this to see if the issue persists?

Thanks.

cuiyanx, 2017-12-29:
Fixed and closed。

closed 2017-12-29 by cuiyanx