intel/zephyr.js #1652: [board] Fail to build with board module

qiaojingx, 2017-11-13:
### Description

This is a regression issue. Building samples and tests of Board API failed for Arduino101 and FRDM-K64F. Other tests which used board api also failed.
 
### Test Code

[test-board.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board.js);
[test-board-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board-manual.js);
[Board.js](https://github.com/01org/zephyr.js/blob/master/samples/Board.js);
[test-spi.js](https://github.com/01org/zephyr.js/blob/master/tests/test-spi.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js);
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js);

### Steps to Reproduction

1. `make JS=samples/Board.js`
2. `make BOARD=frdm_k64f JS=samples/Board.js`
### Actual Result

[image]
 
[image]

### Expected Result

 Build successful without error

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 8248192 - b81ee15 | Arduino101 & FRDM-K64F | Nov 13, 2017 | Fail
master | e369954 | Arduino101 & FRDM-K64F | Nov 13, 2017 | Pass

### Additional Information

brianjjones, 2017-11-13:
I'm working on a fix for this, should be a simple #if check.

brianjjones, 2017-11-13:
#1653 will fix this compiler issue.

qiaojingx, 2017-11-15:
Verified with commit 8fe2e17. This issue is fixed. Close

closed 2017-11-15 by qiaojingx