intel/zephyr.js #1657: [k64f] LEDR,LEDG,LEDB cannot write

qiaojingx, 2017-11-15:
### Description

This is a regression issue. LEDR,LEDG,LEDB on FRDM-K64F cannot write.
 
### Test Code

[ButtonLEDs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDs-k64f.js)
[Blink-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/basics/Blink-k64f.js) 
[test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js)

### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=tests/test-k64f-pins.js `
2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`

### Actual Result

test-k64f-pins.js:
[image]

ButtonLEDs-k64f.js & Blink-k64f.js:
On board led is not lighted.

### Expected Result

test-k64f-pins.js:
No error.

ButtonLEDs-k64f.js & Blink-k64f.js:
On board led blinks.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 8248192 - b81ee15 | FRDM-K64F | Nov 15, 2017 | Fail
master | e369954 | FRDM-K64F | Nov 15, 2017 | Pass
 
### Additional Information

grgustaf, 2017-12-01:
This should be fixed by #1685.

qiaojingx, 2017-12-06:
Verified with commit cbec79d + #1685 , this issue can be fixed.
I will close this bug after patch merged.

closed 2017-12-07 by qiaojingx