intel/zephyr.js #1661: [aio] 'couldn't read from pin 11' on A101

qiaojingx, 2017-11-16:
### Description

When use A1 on Arduino101, it will print `ipm_console0: '52:(ERROR) couldn't read from pin 11' `sometimes.
 
### Test Code

[AIO.js](https://github.com/01org/zephyr.js/blob/master/samples/AIO.js);
 
### Steps to Reproduction

1. `make JS=samples/AIO.js`
2. `make dfu`
### Actual Result

[image]
 
### Expected Result

No error
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | dcf56a9 | Arduino 101 | Nov 16, 2017 | Fail
 
### Additional Information

cuiyanx, 2017-11-16:
On [/tests/test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js):
```
554:(ERROR) 120 ringbuf put errors (last rval=-122)
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 10'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 12'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 10'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 12'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 14'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 10'
ipm_console0: ''
ipm_console0: '52:(ERROR) couldn't read from pin 12'
```

jimmy-huang, 2018-01-10:
@cuiyanx, Is this still reproducible with Zephyr 1.10? I can't reproduce this with AIO.js andtest-a101-pins.js

cuiyanx, 2018-01-11:
@jimmy-huang 
Yes, this issue has be fixed.
@qiaojingx Please closed it.

jimmy-huang, 2018-01-12:
@cuiyanx how long does it take to see the error on AIO.js?

cuiyanx, 2018-01-12:
@jimmy-huang  From the beginning.

jimmy-huang, 2018-01-12:
@cuiyanx ok, i don't see it quite often, at least not immediately, but i do when i run it for a while, I am wondering if this has to do with the sampling rate, can you change the following line in arc/src/arc_aio.c and see if the problem goes away?

// change it from 26 to 30
.sampling_delay = 30;

cuiyanx, 2018-01-15:
@jimmy-huang 
I run this sample about 5 minutes after I change delay to 30, this issue can not reproduce.

jimmy-huang, 2018-01-16:
@cuiyanx ok, i'll submit a patch to update that, thanks for testing.

commit 16533f2d (2018-01-17): [aio] Increase analog sampling rate (#1762)

This will fix the couldn't read from pin error when AIO
is reading too fast.

Fixes #1661