intel/zephyr.js #1663: [docs] The k64f_pins.md doc need to update

cuiyanx, 2017-11-16:
### Description

We support AIO and PWM APIs to K64f board, but [/docs/k64f_pins.md](https://github.com/01org/zephyr.js/blob/master/docs/k64f_pins.md) doc has not be updated.
So, please do it.
 

### Test Code

[/docs/k64f_pins.md](https://github.com/01org/zephyr.js/blob/master/docs/k64f_pins.md)

 

### Steps to Reproduction

 

### Actual Result

No update.
 

### Expected Result

 update

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | dcf56a9 | FRDM-K64F | Nov 16, 2017 | Fail

 

### Additional Information

grgustaf, 2018-01-08:
This file has been eliminated along the module it documented; closing.

closed 2018-01-08 by grgustaf

cuiyanx, 2018-01-09:
Yeah, closed.