intel/zephyr.js #1664: Fail to compile JS file whose sensor number is greater than 2

xiaoyu3x, 2017-11-17:
## Description

Fail to compile JS file when sensor number is 3 or more in it.

## Steps to Reproduce

Build sample [multiserver2.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver2.js)  or [multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js)
`make ROM=256 JS=multiserver.js`

## Actual Result
[image]

## Expected Result

Make Successfully.

## Test Build

Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [dcf56a9](https://github.com/01org/zephyr.js/commit/dcf56a9f0ed511c96abe67e26f4dc502d1212cc4) | Arduino 101 | Nov 17, 2017 | Fail 
master | [5c8b1f1](https://github.com/01org/zephyr.js/commit/5c8b1f1ebe80ce3303a4026dc95f7fbbb93d60b4) | Arduino 101 | Nov 17, 2017 | Pass

jimmy-huang, 2017-11-27:
Yeah, we are just running out with ROM space now, not sure how to tackle this as we are close to the hardware limit on the Arduino 101.

jimmy-huang, 2018-01-24:
@xiaoyu3x With patch #1789, you can further bump ROM_SIZE to 265, with multiserver.js I can set as high as 272, can you re-test?

xiaoyu3x, 2018-01-25:
hi, @jimmy-huang 

This issue isn't produced with PR #1789 when run `make ROM=272 JS=multiserver.js`.

xiaoyu3x, 2018-01-25:
This issue isn't reproduced, so closed it.

closed 2018-01-25 by xiaoyu3x