intel/zephyr.js #1669: [mac] build linux and qemu target failed on Mac

qiaojingx, 2017-11-20:
### Description

This is an regression issue and it is blocked weekly test on MacOS. Building linux and qemu target on Mac failed. 

### Test Code

All samples and tests

### Steps to Reproduction

1. Building linux on MAC by command: `make BOARD=linux`
2. Building qemu on MAC by follow instructions: [building-qemu-and-arduino-101-targets](https://github.com/01org/zephyr.js#building-qemu-and-arduino-101-targets)

### Actual Result

1. Building linux target failed on Mac.
[image]

2. Building qemu target failed on Mac.
[image]

 ### Expected Result

Building successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 16ad2cf - dcf56a9 | qemu & linux | Nov 20, 2017 | Fail
master | f039536 | qemu & linux | Nov 20, 2017 | Pass

### Additional Information

This issue is caused by commit 16ad2cf.

jimmy-huang, 2017-11-20:
this shoud be fixed by #1666

qiaojingx, 2017-11-23:
Verified with commit 58ebd9e. Make linux target succeeds, but make qemu target still fails.
[image]

PR #1683 (2017-11-30): [build] Fix qemu on MacOS
Fixes #1669
commit cbec79db (2017-11-30): [build] Fix qemu on MacOS (#1683)

Fixes #1669
closed 2017-11-30 by grgustaf