intel/zephyr.js #1670: [ws] BUS FAULT occurs when connect and disconnect websocket server for several times.

qiaojingx, 2017-11-20:
### Description

Try to connect and disconnect with websocket server for several times, `BUS FAULT` error will happen.

### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
 
### Steps to Reproduction

 1. `make JS=samples/websockets/WebSocketServer.js BOARD=frdm_k64f`
 2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`
 3. `node samples/websockets/NodeWebSocketClient.js`
 4. CTRL+C to disconnect 
 5. repeat steps 3 and 4(error happens after about 5 times).

### Actual Result

[image]

 ### Expected Result

 No error occurs.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | dcf56a9 | FRDM-K64F | Nov 20, 2017 | Fail
 
### Additional Information

jimmy-huang, 2018-01-18:
@qiaojingx Is this reproducible with master, it seemed the original check was with Zephyr 1.9, but I can't reproduce this on Zephyr 1.10.  However, I do get ip route dropped on the linux time periodically, and I have to add route again.

qiaojingx, 2018-01-19:
Verifed with commit 4b23a47. This issue happens with a much lower reappearance rate now(disconnect for 10+times).
[image]
And if keep connection from the beginning, issue #1725 will occurs.

jimmy-huang, 2018-01-19:
@brianjjones and @jprestwo were able to reproduce this yesterday and root-caused this bug, so a patch is on the way, don't close this until that's in.

closed 2018-01-23 by grgustaf

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b, not reproduced.

commit ba69d2ad (2018-01-23): [timers] Prevent deleting timers / memory twice (#1770)

Fixes #1670