intel/zephyr.js #1679: [board] Build with board module is broken

qiaojingx, 2017-11-27:
### Description

This is a regression issue. Building samples and tests of Board API failed for Arduino101 and FRDM-K64F. Other tests which used board api also failed.
 
### Test Code

[test-board.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board.js);
[test-board-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-board-manual.js);
[Board.js](https://github.com/01org/zephyr.js/blob/master/samples/Board.js);
[test-spi.js](https://github.com/01org/zephyr.js/blob/master/tests/test-spi.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js);
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js);

### Steps to Reproduction

1. `make JS=samples/Board.js`
2. `make BOARD=frdm_k64f JS=samples/Board.js`

### Actual Result

[image]

### Expected Result

 Build successful without error

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 528ec9b - 58ebd9e | Arduino101 & FRDM-K64F | Nov 27, 2017 | Fail
master | 1ebe49b | Arduino101 & FRDM-K64F | Nov 27 2017 | Pass

### Additional Information

grgustaf, 2017-12-01:
Fixed in #1684.

qiaojingx, 2017-12-06:
Verified with commit cbec79d + #1684, this issue can be fixed.
I will close this bug after patch merged.

closed 2017-12-07 by qiaojingx