intel/zephyr.js #1686: [timers] BUS fault error occurs when running test-timers.js on 96b_carbon/stm32f4_disco/olimex_stm32_e407

qiaojingx, 2017-12-04:
### Description

test-timers.js will cause bus fault error on boards: 96b_carbon, stm32f4_disco and olimex_stm32_e407
 
### Test Code

[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)
 
### Steps to Reproduction

96b_carbon:
1. `make BOARD=96b_carbon JS=tests/test-timers.js`
2. `dfu-util -a 0 -d 0483:df11 -D outdir/96b_carbon/zephyr.bin --dfuse-address 0x08000000`

stm32f4_disco:
1. `make BOARD=stm32f4_disco JS=tests/test-timers.js`
2. `cp outdir/stm32f4_disco/zephyr.bin /media/<username>/DIS_F407VG/`

olimex_stm32_e407:
1. `make BOARD=olimex_stm32_e407 JS=tests/test-timers.js`
2. `dfu-util -a 0 -d 0483:df11 -D outdir/olimex_stm32_e407/zephyr.bin --dfuse-address 0x08000000`
### Actual Result

[image]
 
### Expected Result

 
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | cbec79d | 96Boards Carbon | Dec 4, 2017 | Fail
master | cbec79d | STM32F4DISCOVERY | Dec 4, 2017 | Fail
master | cbec79d | OLIMEX-STM32-E407 | Dec 4, 2017 | Fail
 
### Additional Information

jimmy-huang, 2018-01-17:
@qiaojingx can you post the lines of code that it is crashing by searching the instruction address: 8001d72 in the outdir/$BOARD/zephyr/zephyr.lst file.  I've tried reproducing on the 96b_carbon, but not able to, we'll try to reproduce it on other boards as well.

jimmy-huang, 2018-01-17:
I can reproduce this with the old cbec79d commit with Zephyr 1.9, but I can't reproduce this on the olimex_stm32_e407 or 96b_carbon with the latest master with Zephyr 1.10, so I think this issue is fixed with 1.10, please re-test.

qiaojingx, 2018-01-18:
Verified with commit 8b0ac4d. This issue is fixed.
Close

closed 2018-01-18 by qiaojingx