intel/zephyr.js #1687: [linux] Assert doesn't work on Linux

qiaojingx, 2017-12-04:
### Description

Most tests of zephyr.js will get TypeError on Linux.
 
### Test Code

[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js);
[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)
[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)
[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)
 
### Steps to Reproduction

96b_carbon:
1. `make BOARD=96b_carbon JS=tests/test-timers.js`
2. `dfu-util -a 0 -d 0483:df11 -D outdir/96b_carbon/zephyr.bin --dfuse-address 0x08000000`

stm32f4_disco:
1. `make BOARD=stm32f4_disco JS=tests/test-timers.js`
2. `cp outdir/stm32f4_disco/zephyr.bin /media/<username>/DIS_F407VG/`

olimex_stm32_e407:
1. `make BOARD=olimex_stm32_e407 JS=tests/test-timers.js`
2. `dfu-util -a 0 -d 0483:df11 -D outdir/olimex_stm32_e407/zephyr.bin --dfuse-address 0x08000000`
### Actual Result

[image]
 
### Expected Result

 
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | cbec79d | 96Boards Carbon | Dec 4, 2017 | Fail
master | cbec79d | STM32F4DISCOVERY | Dec 4, 2017 | Fail
master | cbec79d | OLIMEX-STM32-E407 | Dec 4, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-12-04:
This was broken by the Modules patch, @brianjjones please take a look at it.

brianjjones, 2017-12-06:
Made a fix for this #1691

qiaojingx, 2017-12-06:
Verified with 79cd6aa from https://github.com/brianjjones/zephyr.js-1/tree/modules-linux. This issue can be fixed. 
I will close this bug after patch merged.

PR #1691 (2017-12-07): [modules] Fix to allow linux to load JS modules
Linux requires the module object be returned to work.
Fixes #1687
closed 2017-12-07 by grgustaf