intel/zephyr.js #1689: [error] Error Type is not subdivided

qiaojingx, 2017-12-05:
### Description

This is a regression issue. When running test-error.js on Arduino101 and FRDM-K64F, two exception tests fails.
 
### Test Code

[test-error.js](https://github.com/01org/zephyr.js/blob/master/tests/test-error.js)
 
### Steps to Reproduction

1. `make JS=tests/test-error.js`
2. `make dfu`

### Actual Result

[image]

### Expected Result

 No FAIL

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | cbec79d | Arduino 101& FRDM-K64F | Nov 4, 2017 | Fail
master | 62697b3 | Arduino 101& FRDM-K64F | Nov 4, 2017 | Pass
 
### Additional Information

This issue is caused by #1658.

cuiyanx, 2017-12-08:
I test this bug with commit 39112c74, this bug has been fixed. Please test again.

qiaojingx, 2017-12-11:
Verified with commit 1b970c8. This issue is fixed.
Close.

closed 2017-12-11 by qiaojingx