intel/zephyr.js #1705: [ble] BUS fault error occurs when connecting on 96b_carbon

cuiyanx, 2017-12-12:
### Description

When I test ble case on 96b_carbon with multiple connection and disconnection. And bus fault error occurs sometimes.
 

### Test Code

[/samples/BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js)

 

### Steps to Reproduction

 

### Actual Result

Bus fault: 
```
BLE sample...
Accepted Connection: 62:54:6b:e1:00:58 (random)
RSSI value: -50dBm
Disconnected Connection: 62:54:6b:e1:00:58 (random)
Accepted Connection: 62:54:6b:e1:00:58 (random)
RSSI value: -50dBm
***** BUS FAULT *****
  Executing thread ID (thread): 0x20000558
  Faulting instruction address:  0x8010e52
  Precise data bus error
  Address: 0x748700
Fatal fault in thread 0x20000558! Aborting.
```

 

### Expected Result

 No error.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c822 | 96b_carbon | Dec 12, 2017 | Fail

 

### Additional Information

jimmy-huang, 2018-01-17:
@cuiyanx did you have to update the bluetooth driver using an external SWD debugger? I just flashed the app and when I scan I do see it showed up:

11:5B:5C:46:1A:43 96boards Test beacon

But I can't connect to it.

cuiyanx, 2018-01-18:
@jimmy-huang Yes, I update the BLE driver with [96b_carbon_nrf51](http://docs.zephyrproject.org/boards/arm/96b_carbon_nrf51/doc/96b_carbon_nrf51.html). If not, you can't connect to 96 board.

cuiyanx, 2018-01-18:
@jimmy-huang 
I test this issue again with ble adapter on master by commit 4b23a47 and get log:
log:
```
BLE sample...
Accepted Connection: 00:19:86:00:0b:0e (public)
RSSI value: -50dBm
Disconnected Connection: 00:19:86:00:0b:0e (public)
Accepted Connection: 00:19:86:00:0b:0e (public)
RSSI value: -50dBm
***** BUS FAULT *****
  Executing thread ID (thread): 0x20000558
  Faulting instruction address:  0x8010e62
  Precise data bus error
  Address: 0x748700
Fatal fault in thread 0x20000558! Aborting.
***** BUS FAULT *****
  Executing thread ID (thread): 0x20001b10
  Faulting instruction address:  0x8010e62
  Precise data bus error
  Address: 0x748700
Fatal fault in thread 0x20001b10! Aborting.
```

jimmy-huang, 2018-01-18:
@cuiyanx did you flash it with one of these? https://github.com/blacksphere/blackmagic/wiki, I think we don't have one laying around so we might have to buy one.

cuiyanx, 2018-01-19:
@jimmy-huang No, I just have [Flyswatter2 JTAG Debugger](http://www.tincantools.com/wiki/Flyswatter2) using Openocd and telnet to flash.