intel/zephyr.js #1707: [ocf] Cannot create multiple resources

qiaojingx, 2017-12-13:
### Description

Error `512:(ERROR) failed to create new resource` will occurs when running ocfserver which has multiple resources.

### Test Code

[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js);
[th02-ocf.js](https://github.com/01org/zephyr.js/tree/master/samples/thermostat/th02-ocf.js);
[test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js);
 
### Steps to Reproduction

1. `make JS=tests/test-ocfserver.js ROM=256`
2. `make dfu`
 
### Actual Result

[image]

 
### Expected Result

Each resource can be created without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | Arduino 101 | Dec 13, 2017 | Fail
master | 1b970c8 | FRDM-K64F | Dec 13, 2017 | Fail
 
### Additional Information

jimmy-huang, 2018-01-11:
This is related to RAM issue on the Arduino101, is this reproducible on the 96b_carbon?

qiaojingx, 2018-01-12:
@jimmy-huang , Ocf cannot be built for 96b_carbon.
Does ocf moudle is supported on 96b_carbon(ble) and OLIMEX STM32 E407(Ethernet)? If so, I need create issues for them.

jimmy-huang, 2018-01-17:
Yeah, it probably make sense to create issues, since they should work, we probably just need to update the json files to add the board for it to compile.

qiaojingx, 2018-01-18:
Ocf issue on 96b_carbon is tracking by #1771 and ocf issue on OLIMEX STM32 E407 is tracking by #1772 .

jimmy-huang, 2018-01-23:
patch submitted #1785, please retest.

cuiyanx, 2018-01-24:
Verified with commit f51fcc3b02 on master.

| | A101 | K64f | 96b |
| :----: | :----: | :----: | :----: |
| [test-ocfserver.js](https://github.com/intel/zephyr.js/blob/master/tests/test-ocfserver.js) | Fail | Fail | Fail |
| [OcfMultiSensor.js](https://github.com/intel/zephyr.js/blob/master/samples/OcfMultiSensor.js) | Fail | Fail | Not Support AIO |

PR #1785 (2018-01-25): [ocf] Fixed iotivity-constrained build
The problem is that all the required preprocessor defines added
to iotivity-constrained using -include doesn't work with CMake,
so instead, we should pass those definitions to cmake using
add_definitions().

Fixes #1707
commit dee11f7a (2018-01-25): [ocf] Fixed iotivity-constrained build (#1785)

The problem is that all the required preprocessor defines added
to iotivity-constrained using -include doesn't work with CMake,
so instead, we should pass those definitions to cmake using
add_definitions().

Fixes #1707
closed 2018-01-25 by grgustaf