intel/zephyr.js #1710: [webusb] No notification pops up for WebUSB.

qiaojingx, 2017-12-13:
### Description

This issue a regression issue. There is no pop up notification when running WebUSB.

### Test Code

[WebUSB.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSB.js);
[WebUSBConfig.js](https://github.com/01org/zephyr.js/tree/master/samples/WebUSBConfig.js);
[WebUSBConfigGroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSBConfigGroveLCD.js);
[test-webusb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-webusb-manual.js);
 
### Steps to Reproduction

1. `make JS=samples/WebUSB.js`
2. `make dfu`
 
### Actual Result

There is no pop up notification when running WebUSB.

 
### Expected Result

A notification pops up.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | Arduino 101 | Dec 13, 2017 | Fail
master | cbec79d | Arduino 101 | Dec 5, 2017 | Fail
 
### Additional Information

brianjjones, 2017-12-28:
Fix is ready for review.  CONFIG_USB_DEVICE_VID and CONFIG_USB_DEVICE_PID have to be added to the .json file for Chrome to display the popup.  I also updated the sample URL to the new one.

qiaojingx, 2018-01-03:
Verified with commit b32969f + #1738 . This issue is not reproduced.
But URL in [WebUSBConfig.js](https://github.com/01org/zephyr.js/tree/master/samples/WebUSBConfig.js), [WebUSBConfigGroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/WebUSBConfigGroveLCD.js), [test-webusb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-webusb-manual.js) need to update, too.

brianjjones, 2018-01-04:
+1 LGTM

closed 2018-01-08 by jimmy-huang

qiaojingx, 2018-01-09:
Verified with commit b17a496， this issue is fixed.

PR #1740 (2018-01-08): [webusb] Update 3 file for webusb popup
Fixed #1710