intel/zephyr.js #1711: [tcp] Stack Check Fail in qemu

qiaojingx, 2017-12-13:
### Description

When running [test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js), [test-tcp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server.js) and [test-tcp4-server-DHCP.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server-DHCP.js) in qemu, `Stack Check Fail` will occur. But not reproduce on A101 and K64F.

### Test Code

[test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js);
[test-tcp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server.js); 
[test-tcp4-server-DHCP.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server-DHCP.js);

### Steps to Reproduction

 1. Start net-tools on Linux:
terminal 1: ` ./loop-socat.sh`
terminal 2: `sudo ./loop-slip-tap.sh`
 2. `make JS=tests/test-tcp6-server.js ROM=256 BOARD=qemu_x86 qemu`

### Actual Result

### Expected Result

No error occurs
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | QEMU | Dec 13, 2017 | Fail

### Additional Information

grgustaf, 2018-01-19:
Fixed in #1776. However I wasn't able to reproduce the bug in the -DHCP case, it just prints one line and I don't remember how to connect to it. So provide more detail if a problem remains.

closed 2018-01-19 by jimmy-huang

qiaojingx, 2018-01-22:
Verified with commit e720399 . This issue is fixed.

PR #1776 (2018-01-19): [qemu] Increase stack size to avoid failure in network tests
Fixes #1711
commit 44589f30 (2018-01-19): [qemu] Remove stack size to avoid failure in network tests (#1776)

Fixes #1711