intel/zephyr.js #1712: [ashell] ashell doesn't work

qiaojingx, 2017-12-13:
### Description

This issue a regression issue. 'Zeph' is outputted instead of commands list when ashell start. Ashell doesn't work.
 
### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`

### Actual Result

[image]

[image]

### Expected Result

commands list of ashell should be outputted via `screen /dev/ttyACM0 115200`.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | Arduino 101 | Dec 13, 2017 | Fail
master | cbec79d | Arduino 101 | Dec 5, 2017 | Pass

### Additional Information

brianjjones, 2017-12-14:
It appears the cmake change broke the command line ashell.  Its rather strange as the IDE version works fine, and you can even use the command line within the console no problem.  Anyway, I'm working on a fix.

brianjjones, 2017-12-16:
OK I've finally found the cause, I've created the fix here #1721

qiaojingx, 2017-12-21:
Verified with commit b32969f. This issue is fixed.
Close.

closed 2017-12-21 by qiaojingx