intel/zephyr.js #1714: [docs] README.md need to update for building on MacOS

qiaojingx, 2017-12-13:
### Description

In order to build on MacOS,  README.md need to update according to http://docs.zephyrproject.org/getting_started/installation_mac.html.
 
### Test Code

[README.md](https://github.com/01org/zephyr.js/blob/master/README.md)
 
### Actual Result

Build with `BOARD=frdm_k64f` will fail with error `Kconfig not found` by following instruction [building-and-running-on-macos](https://github.com/01org/zephyr.js/blob/master/README.md#building-and-running-on-macos).
 
### Expected Result

Build with `BOARD=frdm_k64f` succeed.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | MacOS | Dec 13, 2017 | Fail
 
### Additional Information

execute below commands before build, will fixed `Kconfig not found` issue.

Build Kconfig in $ZEPHYR_BASE/build and add it to path
```
$ cd $ZEPHYR_BASE
$ mkdir build && cd build
$ cmake $ZEPHYR_BASE/scripts
$ make
$ echo "export PATH=$PWD/kconfig:\$PATH" >> $HOME/.zephyrrc
$ source $ZEPHYR_BASE/zephyr-env.sh
```

jimmy-huang, 2018-01-22:
@qiaojingx I've created a patch #1780 to address some of these issues, can you see if anything else needs to be updated?

closed 2018-01-25 by grgustaf

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b. README.md has been updated. Close.

PR #1780 (2018-01-25): [doc] Update Mac instructions for building QEMU
Fixes #1714, #1715
commit 0c0af9b8 (2018-01-25): [doc] Update Mac instructions for building QEMU (#1780)

Fixes #1714, #1715