intel/zephyr.js #1715: [MacOS] Make qemu target failed on MacOS

qiaojingx, 2017-12-13:
### Description

Building qemu target on Mac failed after zephyr updated. 

### Test Code

All samples and tests

### Steps to Reproduction

1. Building qemu on MAC by follow instructions: [building-qemu-and-arduino-101-targets](https://github.com/01org/zephyr.js#building-qemu-and-arduino-101-targets)

### Actual Result

1. Building qemu target failed on Mac, it needs `i586-zephyr-elfiamcu` not `i586-pc-elf`.
I have no idea how to build i586-zephyr-elfiamcu via `ct-ng build`.
[image]

2. Try to change `i586-zephyr-elfiamcu` to `i586-pc-elf` in `deps/zephyr/cmake/toolchain-xtools.cmake`, another error met:
[image]

 ### Expected Result

Building successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1b970c8 | qemu | Dec 13, 2017 | Fail

### Additional Information

jimmy-huang, 2018-01-22:
@qiaojingx I've updated the mac instructions for qemu, can you try to see if you can get it to build following them?  I've ran into many issues on my mac running Sierra, so if you are using a different OS like High Sierra, you may or may not run into the same issues, I've tried to document the work-arounds I did and I am planning to file a bug on Zephyr, you can comment on the patch too if you see any issues so I can update it accordingly.

xiaoyu3x, 2018-01-23:
Verified with new instructions(#1780) for qemu, I can build qemu target successfully.

For arduino 101 target, and i met the below issue by following new instructions. Do you have any work-around for this?
```
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h: Assembler messages:
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:401: Error: unknown opcode 'movl'
make[3]: *** [zephyr/boards/x86/arduino_101/CMakeFiles/boards__x86__arduino_101.dir/pinmux.c.obj] Error 1
make[2]: *** [zephyr/boards/x86/arduino_101/CMakeFiles/boards__x86__arduino_101.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
......
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h: Assembler messages:
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:46: Error: unknown opcode 'pushfl'
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:47: Error: unknown opcode 'cli'
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:48: Error: unknown opcode 'popl'
make[3]: *** [zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj] Error 1
make[2]: *** [zephyr/kernel/CMakeFiles/kernel.dir/all] Error 2
......
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h: Assembler messages:
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:401: Error: unknown opcode 'movl'
/Users/zjs/zephyr.js/deps/zephyr/include/arch/x86/asm_inline_gcc.h:401: Error: unknown opcode 'movl'
[ 29%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/x86/soc/intel_quark/quark_se/power.c.obj
make[3]: *** [zephyr/CMakeFiles/zephyr.dir/arch/x86/soc/intel_quark/quark_se/eoi.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
/Users/zjs/zephyr.js/deps/zephyr/arch/x86/soc/intel_quark/quark_se/power.c: Assembler messages:
/Users/zjs/zephyr.js/deps/zephyr/arch/x86/soc/intel_quark/quark_se/power.c:91: Error: unknown opcode 'sti'
make[3]: *** [zephyr/CMakeFiles/zephyr.dir/arch/x86/soc/intel_quark/quark_se/power.c.obj] Error 1
make[2]: *** [zephyr/CMakeFiles/zephyr.dir/all] Error 2
[100%] Linking C static library ../lib/libjerry-core.a
[100%] Built target jerry-core
[ 18%] Built target jerry-core
Scanning dependencies of target jerry-ext
[ 36%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/arg/arg-transform-functions.c.obj
[ 45%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/arg/arg-js-iterator-helper.c.obj
[ 45%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/arg/arg.c.obj
[ 54%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/module/module.c.obj
[ 63%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/handler/handler-assert.c.obj
[ 72%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/handler/handler-gc.c.obj
[ 81%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/handler/handler-print.c.obj
[ 90%] Building C object jerry-ext/CMakeFiles/jerry-ext.dir/handler/handler-register.c.obj
[100%] Linking C static library ../lib/libjerry-ext.a
[100%] Built target jerry-ext
[ 30%] No install step for 'jerry_project'
[ 31%] Completed 'jerry_project'
[ 31%] Built target jerry_project
make[1]: *** [all] Error 2
make: *** [zephyr] Error 2
```

jimmy-huang, 2018-01-23:
@xiaoyu3x No I don't, did you build with iamcu.config instead of i586.config? Basically, we are just following the instruction here http://docs.zephyrproject.org/getting_started/installation_mac.html, if we can't build native Zephyr apps for A101 with the instructions, we need to file bug on Zephyr so they can fix it.

jimmy-huang, 2018-01-24:
It turns out that the error you are seeing is because it's compiling the arduino_101 with the ARC compiler, not with the crosstool-ng one, since our Makefile builds both ARC and x86 at the same time.  But I tried building just x86 by unsetting the CROSS_TOOL, then I am getting  linker error with gcc, i think it should used clang instead of gcc.  So I filed a bug
https://github.com/zephyrproject-rtos/zephyr/issues/5807, hopefully they fix this, then we can probably update our Makefile to build it correctly.

qiaojingx, 2018-01-24:
Thanks for trying this out.  Can you add the issue ([zephyrproject-rtos/zephyr#5807](zephyrproject-rtos/zephyr#5807)) to arduino 101 building in new instructions(#1780). Then we can close #1715 & #1714 after #1780 merged.

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b. Qemu target can be built by following new instructions. Close

closed 2018-01-25 by qiaojingx