intel/zephyr.js #1725: [ws] Cannot send data to peer(-22) on K64F and QEMU

qiaojingx, 2017-12-18:
### Description

 When running [WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js) and [WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js) on K64F and qemu, data transmission is stuck and `Error: Cannot send data to peer(-22)` is printed after connect with node client for a short while. And, node client cannot connect with WebSocketServer again.

### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js)
 
### Steps to Reproduction

1.  `make JS=samples/websockets/WebSocketServer.js BOARD=frdm_k64f`
2.  `cp outdir/frdm_k64f/zephyr/zephyr.bin /media/<username>/MBED/`
3. Connect with PC via Ethernet cable.
4.  add router and address on Linux
`ip route add 2001:db8::/64 dev <ethX>`
`ip addr add 2001:db8::2 dev <ethX>`
5. connect with node client: 
`node samples/websockets/NodeWebSocketClient.js`

### Actual Result

data transmission is stuck and `Error: Cannot send data to peer(-22)` prints.
[image]
Connection cannot be setup again.

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 616c13f  | FRDM_K64F | Dec 18, 2017 | Fail
master | 616c13f  | qemu | Dec 18, 2017 | Fail
master | 1b970c8  | FRDM_K64F | Dec 11, 2017 | Fail by #1670 
master | 1b970c8  | qemu | Dec 11, 2017 | Pass

### Additional Information

brianjjones, 2018-01-23:
#1770 may have fixed this issue.  I couldn't reproduce it after dozens of re-connects.  Please test again with master and see if it is still an issue.  Thanks.

grgustaf, 2018-01-24:
I added a count to the message output and it happens after a while - once after 500 messages, once after 1000 messages for me. So you have to leave it running for a couple minutes.

cuiyanx, 2018-01-29:
@grgustaf I re-test this issue and get the same error again with d09e4b3 on master.

Log:
```
PONG: PING
MESSAGE: Message #221
MESSAGE: Message #222
MESSAGE: Message #223
MESSAGE: Message #224
MESSAGE: Message #225
MESSAGE: Message #226
MESSAGE: Message #227
MESSAGE: Message #228
MESSAGE: Message #229

249:(ERROR) Cannot send data to peer (-22)
MESSAGE: Message #230
```

qiaojingx, 2018-01-29:
Re-test with commit d09e4b3 in Qemu, got the same error as below:
[image]

grgustaf, 2018-02-01:
@qiaojingx, one thing I realized not too long ago is that net-tools should be updated, where the loop-slip-tap.sh and loop-socat.sh scripts come from. Do you guys keep those up to date? I wonder if that could have anything to do with the difference we see.

pfalcon, 2018-02-01:
Guys, you should know that TCP handling in Zephyr still has enough issues :-I. I specifically write because -22 sounds familiar, it's -EINVAL, I saw it being throw from some networking call where it doesn't make much sense, and one case was recently "fixed": https://github.com/zephyrproject-rtos/zephyr/pull/5864 . Fixed in a sense that instead of that weird error, the proper -ENOMEM is now thrown.

(Bottom line is: definitely look for what may be wrong on your side, but there's enough wrong on Zephyr side too. So, if you can reliably reproduce some failure, please report it upstream. (But the problem is that they aren't really deterministic.))

grgustaf, 2018-02-01:
Yeah I did track that down to somewhere in prepare_segment or sub-call, so that could change things. Thanks. I think the *reason* we were running out of memory though was because the ring buffer was too small and some pending callbacks were getting lost.

One implication of that was if we fail to queue up a callback we may have some rollback work to do to avoid leaking memory. So a flaw in our system is that signal_callback returns void and doesn't let you know when that happened. Glad you reminded me of this because I had forgotten to track that anywhere.

pfalcon, 2018-02-01:
> Yeah I did track that down to somewhere in prepare_segment or sub-call, so that could change things. 

Yeah, that's definitely it.

> So a flaw in our system is that signal_callback returns void and doesn't let you know when that happened.

Yeah, proper error  reporting is important. (Full) recovery may be not always possible, but at least users should get an indication what's wrong, so they could bump callback queue size/net buffer number/etc.

Thanks!

qiaojingx, 2018-02-02:
@grgustaf , `loop-slip-tap.sh` and `loop-socat.sh` scripts come from https://github.com/zephyrproject-rtos/net-tools. I am using commit [3567bc1](https://github.com/zephyrproject-rtos/net-tools/commit/3567bc1d676e0e95f734cb101d98ba113e155438) which is two commits behind the latest master. Thanks for reminding, I will keep these up to date.

qiaojingx, 2018-05-24:
Verified with commit f90d9d1. This issue is not reproduced.
Close.

PR #1793 (2018-01-25): [ws] Fix networking hiccups by increasing ring buffer size
Fixes #1725
commit 4e857693 (2018-01-25): [ws] Fix networking hiccups by increasing ring buffer size

Fixes #1725
closed 2018-05-24 by qiaojingx