intel/zephyr.js #1731: [ambientlight] Pins not found in AmbientLight Sensor

qiaojingx, 2017-12-19:
### Description

This is a regression issue. Pins cannot be analyzed when running [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js) on Ardunio101.

### Test Code

 [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js);
 [test-ambientlight.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ambientlight.js)
 
### Steps to Reproduction

1.  `make JS=samples/AmbientLight.js`
2.  `make dfu`

### Actual Result

[image]

### Expected Result

The value of AmbientLight sensor should output without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 616c13f  | Arduino101 | Dec 19, 2017 | Fail
master | 1b970c8  | Arduino101 | Dec 12, 2017 | Pass

### Additional Information

grgustaf, 2018-01-08:
@jimmy-huang it looks like your AmbientLight code probably needs to change to deal with the new pin naming? I might need to help?

jimmy-huang, 2018-01-08:
Yeah, does look like it needs to be updated, go ahead if you find a quick fix, or I can look at it later

qiaojingx, 2018-01-10:
Verified with commit 03f6e1c + #1746. This issue is fixed.
But AmbientLightSensor is still not working on FRDM-K64F after PR #1746.
```
Ambient light test...

110:(ERROR) failed to open dev
Error: zjs_sensor_create failed to init
```

jimmy-huang, 2018-01-11:
Yeah, it seems like frdm_k64f is not yet ready to pull in ADC to process the ambient light sensor as it will also need to pull in math library to do the conversions, these sensors should be using the grove light driver in Zephyr, but I just checked the Zephyr Grove Light samples, it doesn't allow you to dyanmically change the AIO pin via API.  So I think this patch should just focus on fixing the A101 issue, and we'll have to add support for frdm_k64f in a later patch.

qiaojingx, 2018-01-11:
OK, then I will close this issue after #1746 merge.

PR #1746 (2018-01-11): [sensor] Fixed ambient light sensor pin lookup
Fixes #1731
commit 4bced57f (2018-01-11): [sensor] Fixed ambient light sensor pin lookup (#1746)

Fixes #1731
closed 2018-01-11 by grgustaf