intel/zephyr.js #1735: [tcp] BUS Fault occurs when running TCPEchoServ4_DHCP.js over night

qiaojingx, 2017-12-28:
### Description

Using python client to send data continuously to TCPEchoServ4_DHCP.js on FRDM-K64F over night, BUS Fault error will occurs.

### Test Code

 [TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samplesTCPEchoServ4_DHCP.js)
Python client: test-tcp4-client-dhcp.py
```
 # !usr/bin/python
# coding:utf-8

import time
import socket
import sys

def main():
    print "Socket client creat successful"

    host = sys.argv[1]
    port = 4242
    bufSize = 1024
    addr = (host, port)
    Timeout = 300

    mySocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    mySocket.settimeout(Timeout)
    try:
        mySocket.connect(addr)
    except :
        print "Socket connect failed"

    try:
        sendData = "hello"
        mySocket.sendall(sendData)
        print "Send data: ", sendData
    except :
        print "Socket send data failed"

    try:
        recvData = mySocket.recv(bufSize)
        recvData = recvData.strip()
        print "Got data: ", recvData
    except :
        print "Socket receive data failed"

    if recvData == sendData:
        mySocket.close()
        print "Close socket"
    else :
        print "Data error"

if __name__ == "__main__" :
    i = 0
    while 1:
        try:
            i = i + 1
            print "\nTimes: ", i
            main()
            time.sleep(10)
        except KeyboardInterrupt :
            print "exit python case"
            break
        except :
            print "Error occurred"
            break
```
### Steps to Reproduction

1.  `make JS=samples/TCPEchoServ4_DHCP.js BOARD=frdm_k64f`
2.  `cp outdir/frdm_k64f/zephyr/zephyr.bin /media/<username>/MBED/`
3.  connect FRDM-K64F and PC to the same router
4. `python test-tcp4-client-dhcp.py <IP address>`

### Actual Result

[image]

### Expected Result

No error occurs
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master |  b32969f   | FRDM-K64F | Dec 28, 2017 | Fail

### Additional Information

jimmy-huang, 2018-02-01:
@cuiyanx please re-test with #1816

qiaojingx, 2018-02-01:
Verified with commit d09e4b3 + #1816，this issue is not reproduced after 4 hours.

closed 2018-02-01 by jimmy-huang

qiaojingx, 2018-02-02:
Verified with commit d09e4b3 + #1816，this issue is not reproduced after running over night.

commit 5182eb0e (2018-02-01): [events] Fix mem leak of overwriting callback id (#1816)

Only one event callback was supposed to be created, but the line was
moved to zjs_make_emitter so that it was created for each new emitter
object. For tests that close and open sockets repeatedly, for example,
this eventually caused the system to run out of memory.

Fixes #1735, #1736, #1737