intel/zephyr.js #1736: [ocf/udp/tcp/ws] Server on Arduino101 over ENC28J60 will hang after long-running

qiaojingx, 2017-12-28:
### Description

OcfServer, UDPServer,TCPServer and WebSocketServer all hang after long-running on Arduino101 with ENC28J60 Ethernet Board. 
 
### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js);
[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js);
[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);

### Steps to Reproduction

1. connect ENC28J60 to Arduino101 by wiring the below pins on boards.
```
CS  -> IO10
SI -> IO11
SO -> IO12
SCK -> IO13
VCC -> 3.3v
GND -> GND
INT -> IO4
```
2. Connect ENC28J60 Ethernet Board to PC via Ethernet cable.
3. Flash Arduino101 with test sample.
```
$ make JS=samples/UDPEchoServ4.js FORCE=zjs_net_l2_enc28j60.json
$ make dfu
```
4. add router and address to PC
```
ip route add 192.0.2.0/24 dev eno1
ip addr add 192.0.2.2 dev eno1
```
5. Connect UDPEchoServ4.js using python client: 
`python test-udp4-client.py`
 
### Actual Result

1. OcfServer.js
OcfServer hangs and OcfClient got error `[FATAL]:10`, cannot reconnect again.
[image]
2. WebSocketServer.js
WebSocketServer hangs, cannot receive data from node client, cannot reconnect with node client again.

3. TCPEchoServ6.js
TCPServer6 got error "malloc failed" first, then hangs. cannot receive data from python client, cannot reconnect with python client again.
[image]

4. UDPEchoServ4.js
UDPServer4 hangs, cannot receive data from python client, cannot reconnect with python client again.

### Expected Result

Server should always works fine.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master |  b32969f | Arduino 101 | Dec 28, 2017 | Fail
 
### Additional Information

The matched server and client are here: https://github.com/01org/web-test-suite/tree/master/zephyr.js/tools/tests

cuiyanx, 2018-01-26:
I re-test this issue and get another issue that looks like #1737 with ENC28J60 Ethernet Board.

Server log([TCPEchoServ6.js](https://github.com/intel/zephyr.js/blob/master/samples/TCPEchoServ6.js)):
```
connection made
port: 4242
family: IPv6
address: a00:0:2001:db8::
socket.localPort: 4242
socket.localAddress: a00:0:2001:db8::
socket.remotePort: 4242
socket.remoteAddress: a00:9276:2001:db8::
socket.remoteFamily: IPv6
Got data: hello
Error: wrong type of argument
***** CPU exception 6
Current thread ID = 0xa8005898
Faulting segment:address = 0x0008:0xb0000008
eax: 0x00000000, ebx: 0x0000c054, ecx: 0xa800575c, edx: 0x00000000
esi: 0x00000354, edi: 0x00000006, ebp: 0xaaaaaaaa, esp: 0xa8005730
eflags: 0x10246
Fatal fault in ISR! Spinning...

```

Client log([test-tcp6-client.py](https://github.com/01org/web-test-suite/blob/master/zephyr.js/tools/tests/test-tcp6-client.py)):
```
Times:  450
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  451
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  452
Socket client creat successful
Socket connect failed
Socket send data failed
Got data:  
Data error

```

jimmy-huang, 2018-01-26:
See my comment on #1737, I think we have a leak somewhere, because it was failing on malloc

grgustaf, 2018-02-01:
As @jimmy-huang suggested in IRC, it seems like this may have been fixed by #1816 as well.

jimmy-huang, 2018-02-01:
@cuiyanx please re-test with #1816

cuiyanx, 2018-02-02:
I re-test this issue with PR #1816, this issue is fixed and not reproduced. So closed.

closed 2018-02-02 by cuiyanx