intel/zephyr.js #1737: [tcp] `General Protection Fault` occurs when running TCPEchoServ6.js on arduino101 over ble for long time

qiaojingx, 2017-12-28:
### Description

`General Protection Fault` occurs after running TCPEchoServ6.js for 1 hour on arduino101 over ble. 
 
### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js);

### Steps to Reproduction

1. Flash Arduino101 with test sample.
```
$ make JS=samples/TCPEchoServ6.js ROM=256
$ make dfu
```
2. connect to PC via 6lowpan, add router and address.
```
modprobe bluetooth_6lowpan
echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
echo "connect FF:EE:DD:CC:BB:AA 2" > /sys/kernel/debug/bluetooth/6lowpan_control
ip -6 route add 2001:db8::/64 dev bt0
ip -6 addr add 2001:db8::2 dev bt0
```
3. Connect TCPEchoServ6.js  using python client: 
`python test-tcp6-client.py`
 
### Actual Result

`General Protection Fault` occurs after about 1hour.
[image]

### Expected Result

Server should always works fine.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master |  b32969f | Arduino 101 | Dec 28, 2017 | Fail
 
### Additional Information

The matched python client is here: https://github.com/01org/web-test-suite/tree/master/zephyr.js/tools/tests

jimmy-huang, 2018-01-25:
Is this still reproducible? Right after I do After I do:
ip -6 route add 2001:db8::/64 dev bt0

I could still ping the board, but then after I do

ip -6 addr add 2001:db8::2 dev bt0, i can't no longer ping the board.

and when I run test-tcp6-client.py, it just sits there, not connecting.  @qiaojingx are you able to connect?

cuiyanx, 2018-01-26:
@jimmy-huang 
I re-test this issue, and get the same result.

Server log(a101):
```
connection made
port: 4242
family: IPv6
address: a00:0:2001:db8::
socket.localPort: 4242
socket.localAddress: a00:0:2001:db8::
socket.remotePort: 4242
socket.remoteAddress: a00:84c2:2001:db8::
socket.remoteFamily: IPv6
Got data: hello
Error: wrong type of argument
***** General Protection Fault
***** Exception code: 0x2a34
Current thread ID = 0xa80059e0
Faulting segment:address = 0x0008:0xa800760e
eax: 0x00000000, ebx: 0x00000472, ecx: 0xa8007556, edx: 0xa8007e37
esi: 0x00000272, edi: 0x00000006, ebp: 0xa800d968, esp: 0xa800d92c
eflags: 0x10246
Fatal fault in essential thread! Spinning...
```
Client log(Ubuntu 16.04):
```
Times:  354
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  355
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  356
Socket client creat successful
Send data:  hello
Socket receive data failed
Error occurred

```

jimmy-huang, 2018-01-26:
I still can't get it to connect somehow:
$ python ./tests/tools/test-tcp6-client.py 
Socket client creat successful
Traceback (most recent call last):
  File "./tests/tools/test-tcp6-client.py", line 44, in <module>
    main()
  File "./tests/tools/test-tcp6-client.py", line 18, in main
    mySocket.connect(addr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 111] Connection refused

cuiyanx, 2018-01-26:
@jimmy-huang 
Do you use [TCPEchoServ6.js](https://github.com/intel/zephyr.js/blob/master/samples/TCPEchoServ6.js) as server and [test-tcp6-client.py](https://github.com/01org/web-test-suite/blob/master/zephyr.js/tools/tests/test-tcp6-client.py) as client? I think you had some mistake for client case, because port of the sample is 4242, but port of the test is 9876.

cuiyanx, 2018-01-26:
@jimmy-huang 
TCP6 client src:
```
# !usr/bin/python
# coding:utf-8

import time
import socket

def main():
    print "Socket client creat successful"

    host = "2001:db8::1"
    port = 4242
    bufSize = 1024
    addr = (host, port)
    Timeout = 300

    mySocket = socket.socket(socket.AF_INET6, socket.SOCK_STREAM)
    mySocket.settimeout(Timeout)
    try:
        mySocket.connect(addr)
    except :
        print "Socket connect failed"

    try:
        sendData = "hello"
        mySocket.sendall(sendData)
        print "Send data: ", sendData
    except :
        print "Socket send data failed"

    try:
        recvData = mySocket.recv(bufSize)
        recvData = recvData.strip()
        print "Got data: ", recvData
    except :
        print "Socket receive data failed"

    if recvData == sendData:
        mySocket.close()
        print "Close socket"
    else :
        print "Data error"

if __name__ == "__main__" :
    i = 0
    while 1:
        try:
            i = i + 1
            print "\nTimes: ", i
            main()
            time.sleep(10)
        except KeyboardInterrupt :
            print "exit python case"
            break
        except :
            print "Error occurred"
            break
```

jimmy-huang, 2018-01-26:
@cuiyanx ok, so the client is not available in the repo, is this part of the another test-suite?  Because the tests/tools/test-tcp6-client.py looks totally different.

jimmy-huang, 2018-01-26:
Ok, I am able to reproduce it now with your script, and I also happen to crash on 356th iteration, so I don't think that's a coincidence:

Times:  356
Socket client creat successful
Send data:  hello
Socket receive data failed
Error occurred

jimmy-huang, 2018-01-26:
So it looks like we have a memory leak somewhere:

0:2001:db8::
socket.remotePort: 4242
socket.remoteAddress: a00:b0dc:2001:db8::
socket.remoteFamily: IPv6
received data, context=0xa8004e00, data=0xa800bf60, len=5
trace: T:0xa8005a60: create object: 0xa80083eb
emitting event 'data'
Got data: hello
Sent 5 bytes
Sent 5 bytes
Sent 0 bytes
closing socket, context=0xa8004e00, handle=0xa8013d00
socket=0xa80082ab
queuing event 'close'
callback ID 4294967295 does not exist
connection made, context 0xa8004e68 error 0
deferring work: 32 bytes
trace: T:0xa8005a60: create object: 0xa8008413
malloc(1984) failed

cuiyanx, 2018-01-29:
@jimmy-huang Yes, We have some test cases for stability test. They are TCP6 server, TCP6 client, UDP6 client and WebUSB for ZJS.

cuiyanx, 2018-02-01:
I re-test this issue with PR #1816, this issue is fixed and not reproduced.

Client log:
```
Times:  355
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  356
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  357
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

Times:  358
Socket client creat successful
Send data:  hello
Got data:  hello
Close socket

```

PR #1816 (2018-02-01): [events] Fix mem leak of overwriting callback id
Only one event callback was supposed to be created, but the line was
moved to zjs_make_emitter so that it was created for each new emitter
object. For tests that close and open sockets repeatedly, for example,
this eventually caused the system to run out of memory.

Fixes #1737
closed 2018-02-01 by jimmy-huang