intel/zephyr.js #1743: [gfx] If 'x0 = x1, y0 > y1' or 'x0 > x1, y0 = y1', that line drawing is different of expectation.

cuiyanx, 2018-01-09:
### Description

When I test drawLine() function, I find that line drawing is different of expectation if I set 'x0 = x1, y0 > y1' or 'x0 > x1, y0 = y1'. The direction of the line is wrong.
 

### Test Code

Simple test:
```
var LCD = require("ST7735.js");
var board = require("board");
var drawImmediate = board.name === "arduino_101" ? true : false;
var gfxLib = require("gfx");

var GFX = gfxLib.init(LCD.width, LCD.height, LCD.initScreen, LCD.drawCB, drawImmediate, LCD);

var BLACK =  [0x00, 0x00];
var WHITE =  [0xFF, 0xFF];

GFX.fillRect(0, 0, 128, 160, BLACK);

GFX.drawLine(64, 80, 64, 10, WHITE);
GFX.drawLine(64, 80, 10, 80, WHITE);
```

 

### Steps to Reproduction

 

### Actual Result

Line1: Starting point： [64, 80], End point： [64, 10]
Line2: Starting point： [64, 80], End point： [10, 80]

### Expected Result

Line1: Starting point： [64, 80], End point： [64, 150]
Line2: Starting point： [64, 80], End point： [118, 80]
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b17a496b0 | Arduino 101 | Jan 9, 2018 | Fail

 

### Additional Information

qiaojingx, 2018-01-16:
Verified with commit 8b0ac4d + #1755. This issue is not reproducible.

closed 2018-01-22 by grgustaf

cuiyanx, 2018-01-23:
Fixed and closed

PR #1755 (2018-01-22): [gfx] Draw a line correctly in special cases
Fixed #1743
commit bda14eff (2018-01-22): [gfx] Draw a line correctly in special cases (#1755)

Fixes #1743