intel/zephyr.js #1744: [gfx] When I set coordinate as negative number, the coordinate has be fixed to zero.

cuiyanx, 2018-01-09:
### Description

When I set coordinate as negative number, the coordinate has be fixed to zero.
 

### Test Code

Simple test:
```
var LCD = require("ST7735.js");
var board = require("board");
var drawImmediate = board.name === "arduino_101" ? true : false;
var gfxLib = require("gfx");

var GFX = gfxLib.init(LCD.width, LCD.height, LCD.initScreen, LCD.drawCB, drawImmediate, LCD);

var BLACK =  [0x00, 0x00];
var WHITE =  [0xFF, 0xFF];

GFX.fillRect(0, 0, 128, 160, BLACK);

GFX.drawLine(-50, 30, -90, 100, WHITE);
GFX.drawLine(30, -50, 100, -90, WHITE);
```

 

### Steps to Reproduction

 

### Actual Result

Line1: Starting point： [0, 30], End point： [0, 100]
Line2: Starting point： [30, 0], End point： [100, 0]
 

### Expected Result

Outside of screen and don't display.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | b17a496 | Arduino 101 | Jan 9, 2018 | Fail

 

### Additional Information

grgustaf, 2018-01-09:
Hmm, yeah really doing this right should probably draw any on-screen portion of the line and drop the rest, e.g. if you chose (-50, 30, 40, 100), whereas it appears we're clamping to on-screen 0 which really draws the wrong line.

qiaojingx, 2018-01-16:
Verified with commit 8b0ac4d + #1761 . This issue is not reproducible.

brianjjones, 2018-01-23:
With @cuiyanx PR's this should no longer be an issue, it can be closed.

closed 2018-01-26 by grgustaf

cuiyanx, 2018-01-29:
Fixed and closed.

PR #1761 (2018-01-26): [gfx] Can set coordinate as negative number
Fixed #1744