intel/zephyr.js #1759: I'd like to fix the WebIDL in the documentation -- thoughts?

t-harvey, 2018-01-12:
Hi all,
I'm writing a WebIDL-->Jerryscript compiler/code-generator, and I've been using the excellent object-model that zephyr.js has built by copy/pasting the WebIDL out of the documentation and using it as a test bed for my project.

Unfortunately, the WebIDL is not strictly conforming to the spec, and I would like to take on the task of updating the documentation with syntactically correct WebIDL -- before I put in a pull request with my changes, I'd like to see if anyone has any strong feelings that I shouldn't?

Thanks very much,
Tim

jimmy-huang, 2018-01-12:
Yes, that would be great, anything to help make our spec WebIDL compliant is always welcomed, go ahead and submit the patch and we can review them :+1:

grgustaf, 2018-01-12:
One caveat is just yesterday I removed the WebIDL from buffer.md, in favor of a style more like what Node.js does for documentation. But you're welcome to reintroduce it if you want to correct it.

Alternately, maybe we could move it into a separate file like buffer.idl and link to that in buffer.md; I'm up for suggestions. As you see we've just been using it in a kind of ad hoc manner.

t-harvey, 2018-01-28:
Guys,
I just created a pull request (#1800) to let people (you guys? :-) examine the kinds of changes I'd like to make to the documentation.