intel/zephyr.js #1771: [ocf] Build ocf module for 96b_carbon failed

qiaojingx, 2018-01-18:
### Description

 Unable to build ocf module for 96b_carbon board.

### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js)
 
### Steps to Reproduction

`1. make BOARD=96b_carbon JS=samples/OcfServer.js `

### Actual Result

Build failed.
[image]

### Expected Result

Build succeed. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4b23a47 | 96b_carbon | Jan 18, 2018 | Fail
 
### Additional Information

qiaojingx, 2018-01-18:
Verified with commit 4b23a47 + #1768. Ocf module can be built for 96b_carbon. But Ocf connection can't be established. The same phonmenon as it is on olimex_stm32_e407(#1772)

jimmy-huang, 2018-01-22:
The build issue has been fixed, so we'll track the connect issue with #1772.

closed 2018-01-22 by jimmy-huang

qiaojingx, 2018-01-26:
Verified with commit 0c0af9b in master. OCF works fine on 96b_carbon board.