intel/zephyr.js #1773: FYI: Zephyr SDK 0.9.2 arm-gcc is broken for Cortex-M0 (hit by JerryScript)

pfalcon, 2018-01-18:
Seeing you guys are working on more boards, wanted to share this issue: https://github.com/zephyrproject-rtos/meta-zephyr-sdk/issues/91 .

So, if by any change you have Cortex-M0 boards in your queue, feel free to put them at the back of it. (An example would include nRF51 (nRF52 is Cortex-M3)).

Just FYI, decided to submit this just in case. Feel free to close.

grgustaf, 2018-01-18:
Thanks for the tip!

grgustaf, 2018-01-19:
I didn't like how long it took me to figure out if this was an immediate issue for us, so I just created a [tested boards](https://github.com/grgustaf/zephyr.js-1/blob/a236cb657fb714e69653068b0fd6f04ffd538403/docs/boards/summary.md) summary table in #1775.

We haven't yet tested on any Cortex-M0 boards. I wonder if their specs tend to be too low to be able to run ZJS? Anyway, we can probably close this soon but I'll leave it open for now to give others a chance to notice it.

pfalcon, 2018-01-19:
Yes, sorry that I didn't clarify that. From my cursory checking, this didn't seem like an immediate issue. At the same time, as you guys start to support more boards, and community might look into that too, it may pop up "any time now", so I just wanted to share a forward notice to avoid any possible confusion/double-work.

> We haven't yet tested on any Cortex-M0 boards. I wonder if their specs tend to be too low to be able to run ZJS?

Well, it depends. Cortex-M0 are simpler and lower-clock, thus lower-power. Commonly used with wireless devices. ROM/RAM amounts vary, e.g. FRDM-KW41Z board for which the issue was discovered, has 512K/128K, which is pretty beefy.

> Anyway, we can probably close this soon 

Sure. I had a choice of dumping an offtopic note in an existing ticket or creating a new note-ticket, just decided to do the latter ;-).