intel/zephyr.js #1787: WebSocketServer samples shows two PONG events when there should be one

grgustaf, 2018-01-24:
Run the WebSocketServer.js sample on K64F at least, then connect with node client. In between the 10 message lines, it shows two PONG prints. There should only be one.

Investigating a little yesterday, it appeared there's only one `tcp_received` call and one `zjs_defer_work` call but somehow it's resulting in two pong events. I wonder if maybe the message isn't getting pulled out of the ring buffer the first time for some reason.

brianjjones, 2018-01-26:
FWIW the second pong is coming from the client side, which is being run by node - and running their version of WS.  So its quite possible it is a node bug or Zephyr issue.

brianjjones, 2018-01-26:
Confirmed, Node.js automatically sends a pong in response to a ping.  Having the ws.pong line in there causes two pongs to be sent.  See #1797

@grgustaf should I update our WS code to match this behavior? I assume this would go in after 0.5.

commit 53ab9e7d (2018-01-26): [ws] Removing unneeded pong (#1797)

Node.js automatically returns a pong when it gets a ping,
calling ws.pong causes two to be sent to the webserver.

Fixes #1787
closed 2018-01-26 by grgustaf