intel/zephyr.js #1801: [build] When building with Linux target, I have got some "warning".

cuiyanx, 2018-01-29:
### Description

When I am building with Linux target, I have got some "warning". Although I can finished the Linux building, but please fix it.
 

### Test Code

 

### Steps to Reproduction

` make BOARD=linux`

### Actual Result

Warning 1:
`warning: "OC_CLOCK_CONF_TICKS_PER_SECOND" redefined`
 
Warning 2:
`warning: variable ‘key’ set but not used [-Wunused-but-set-variable]`

### Expected Result

No warning.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
:---:|:---:|:---:|:---:|:---:
master | d09e4b36 | Linux | Jan 29, 2018 | Fail

 

### Additional Information

grgustaf, 2018-01-30:
I don't seem to reproduce this. Can you give more context around the warnings? Are you sure there wasn't more to the build command line?

jimmy-huang, 2018-01-30:
I do see the warning: variable ‘key’ set but not used [-Wunused-but-set-variable] warning, but i don't see the first one.

grgustaf, 2018-01-30:
OK I do see that first error in Travis output, so we should be able to address this.

jimmy-huang, 2018-01-30:
patch submitted #1803

cuiyanx, 2018-01-30:
I re-test this issue with #1802 and #1803, this issue will be fixed.
Wait for merge #1802 and #1803, I will close this.

PR #1803 (2018-02-01): [ocf] Fixed compile warnings on linux
Fixes #1801
commit 1edf34c5 (2018-02-01): [ocf] Fixed compile warnings on linux (#1803)

Fixes #1801
commit 4cf09c54 (2018-02-01): [ocf] Fixed compile warnings on linux (#1803)

Fixes #1801
PR #1802 (2018-02-01): [callbacks] Eliminate compile warning
Fixes #1801
closed 2018-02-01 by grgustaf