intel/zephyr.js #1804: [ocf] Cannot get return value when first retrieve from a multiple resources Server on K64F

qiaojingx, 2018-01-30:
### Description

when first retrieve from a multiple resources ocfserver on K64F, no sensor value get. This issue is not reproduced on Arduino101.
 
### Test Code

[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js)

OcfMultiClient.js code:
```
var ocf = require('ocf');
var client = ocf.client;
var t1 = null;
ocf.start();

t1 = setInterval(function() {
client.findResources({ resourceType:"core.light" }).then(function(resource) {
    console.log("findResources 'core.light' was successful, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("light: " + res.properties.light);
    }, function(error) {
    });
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
});

setTimeout(function() {
    client.findResources({ resourceType:"core.temp" }).then(function(resource) {
        console.log("findResources 'core.temp' was successful, deviceId=" + resource.deviceId);

        client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
            console.log("temp: " + res.properties.temp);
        }, function(error) {
        });
    }, function(error) {
        console.log("findResources() returned an error: " + error.name);
    });
}, 5000);
}, 10000); 
```
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/OcfMultiSensor.js`
2. `cp outdir/frdm_k64f/zephyr/zephyr.bin /media/<username>/MBED/`
3. connect PC via Ethernet cable.
4. add router and address on PC
```
ip -6 route add 2001:db8::/64 dev eno1
ip addr add 2001:db8::2 dev eno1
```
5. `./outdir/linux/release/jslinux ../../OcfMultiClient.js --noexit`

### Actual Result

No sensor value is returned at the beginning.
[image] 

### Expected Result

Can get sensor value from the beginning.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | d09e4b3 | FRDM_K64F | Jan 31, 2018 | Fail
master | d09e4b3 | Arduino101 | Jan 31, 2018 | Pass

### Additional Information

jimmy-huang, 2018-01-30:
I was able to see the res from k64f if I run OcfMultiServer.js and then OcfClient.js:

jimmy@jimmy-dev:~/projects/zephyrjs/zephyr.js (update-doc)$ ./outdir/linux/release/jslinux samples/OcfClient.js --noexit

Started OCF client
findResources() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e
retrieve() was successful, deviceId=ocf://c9920b68-2ab7-4e7e-5169-346b073c448e

There's no OcfMultiClient.js file, where is this client app?

qiaojingx, 2018-01-31:
Yes, there is no OcfMultiClient.js file in zephyr.js, we created it ourselves for OcfMultiSensor.js testing. The code is in **Test Code** Section above.