intel/zephyr.js #1809: what is wrong with my arduino101?

qiaojingx, 2018-01-31:


jimmy-huang, 2018-01-31:
have you tried building native zephyr apps and flash them to both x86 and ARC processors separately, and see if you can get that up?  For example, if you build a the helloworld app on ARC, and then build the samples/booting/stub app and flash that the x86.  At least then it rules out if you have a working board or not.  If not, you might try to use the intel A101 recovery tools to flash the original firmware back and then re-flash with Zephyr.

qiaojingx, 2018-02-01:
I have tried with native zephyr apps, it seems the board is working.
[image]

BTW, this issue is not reproduced when running samples in normal ZJS ways.  Only bad in Ashell.
[image]

grgustaf, 2018-02-01:
Yeah, that is interesting that the text line doesn't show up. That does make it seem rather like the ARC image failed to get built or flashed correctly. Possibly it could have to do with the recent changes to ARC build -Os? The version of the ARC image used for ashell probably has more features enabled than usually in your "production mode" builds. I think @jimmy-huang just left but I'll try to talk to him about this tomorrow.

grgustaf, 2018-02-01:
Do non-ARC apps like just simple Blink.js work OK from ashell?

qiaojingx, 2018-02-01:
Blink.js works fine in Ashell.

jimmy-huang, 2018-02-01:
You can flash commit dee11f7abd7ea5c6b4c51f10f8c102b548776385 which is prior to the -Os changes and see if you can get ARC working again, if that works, we might need to file bug againist Zephyr because apparently they didn't fix the issue.

qiaojingx, 2018-02-01:
Re-test with commit d09e4b3. This issue is still there.

jimmy-huang, 2018-02-01:
maybe you can find out which component in ashell that's causing it to not boot, so if you edit src/zjs_ashell.json, and start removing individual components like aio, i2c, or sensor etc, i think these are the only ones that uses the ARC.  At least this narrows down to which thing is borking your image.

jimmy-huang, 2018-02-02:
another thing you can try is the erase the whole flash filesystem on the a101 using ashell in the cmd prompt, to remove any thing on the FS that could cause the board to boot differently.