intel/zephyr.js #1826: The existing file disappears after rebooting or reseting STM32F407G-DISC1 and Olimex STM32-E407 boards

xiaoyu3x, 2018-02-05:
## Issue description
The existing file disappears after rebooting or reseting STM32F407G-DISC1 and Olimex STM32-E407 boards

## Steps to reproduce
1. Build and flash image into board.
2. Start WebIDE and connect board.
3. Save any file into boards( or use 'load' command to create file in console.)
4. Execute 'reboot' command in console or press 'reset' button on board.

## Actual Result

The existing file disappears, e.g. There are "f.js" and "parse.js", but they disappear after reboot or reset board.
[image]

## Expected Result
Files didn't disappeared after reboot or reset board.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [ede4509](https://github.com/intel/zephyr.js/commit/ede4509d98da5f0187b82a7867fc0277641c4ba7) | Arduino 101 | Feb 5, 2018 | Fail

jimmy-huang, 2018-02-12:
This is probably because both the boards do not have a dedicated flash for the filesystem like the A101.  So it's similar to the k64f, where it uses the RAM to simulate the filesystem, therefore it will be gone when you reboot, I don't think we can fix this unless the board can use another filesystem format like SD card support.

xiaoyu3x, 2018-02-13:
Thanks, @jimmy-huang 

According to your comment, I will close it.

closed 2018-02-13 by xiaoyu3x