intel/zephyr.js #1827: "erase" command can't be found in the console of WebIDE with STM32F407G-DISC1 and Olimex STM32-E407 boards

xiaoyu3x, 2018-02-05:
## Issue description
"erase" command can't be found when running 'erase' on the console of WebIDE with STM32F407G-DISC1 and Olimex STM32-E407 boards

## Steps to reproduce
1. Build and flash image into board.
2. Start WebIDE and connect board.
3. Execute 'erase' command in console.

## Actual Result

"erase: command not found" is outputed.
[image]

## Expected Result
Command 'erase' is used normally.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [ede4509](https://github.com/intel/zephyr.js/commit/ede4509d98da5f0187b82a7867fc0277641c4ba7) | Arduino 101 | Feb 5, 2018 | Fail

jimmy-huang, 2018-02-12:
see comment on #1827, only the A101 has a dedicated flash filesystem, in which this erase feature is A101 specific.  All the other boards uses RAM to simulate the FileSystem, so you can't perform erase on these boards.

xiaoyu3x, 2018-02-13:
Thanks, @jimmy-huang 

I think you are right, so I will close it.

closed 2018-02-13 by xiaoyu3x