intel/zephyr.js #1829: Follow new Generic Sensor API

kenchris, 2018-02-06:
The current implementation differs from that of the spec, like onchange is now onreading.

It would be nice to follow the spec now, especially now that FitBit also follows it: https://dev.fitbit.com/build/reference/device-api/sensors/

jimmy-huang, 2018-02-08:
@kenchris Are the readings still a property of the sensor object itself? Or is now part of the event obj when the listener is called?  Because in the fitbit api, there's a AccelerometerReading interface where the readings contains x, y, z but in the generic w3c spec, the example usess sensor.latitude, sensor.longitude, so I am assuming you still retrieve it via sensor.x, sensor.y, sensor.z for the Accelerometer?

kenchris, 2018-02-08:
In the spec they are still part of the object itself, which is the case because of GC issues, I remember.

But onchange is called onreading and the state is not exposed anymore

kenchris, 2018-02-08:
I see now that the FitBit probably implemented this before we made that change, but they do use the right method names

jimmy-huang, 2018-02-08:
ok, thanks, so at least this makes it so the change not so bad

PR #1837 (2018-03-06): [sensor] Implements new W3C Generic Sensor API
In the latest W3C spec, onchange event handler has been renamed to
onreading, and the state property is no longer exposed in the sensor
object.

Fixes #1829
commit 5d277d63 (2018-03-06): [sensor] Implements new W3C Generic Sensor API (#1837)

In the latest W3C spec, onchange event handler has been renamed to
onreading, and the state property is no longer exposed in the sensor
object.

Fixes #1829
closed 2018-03-06 by grgustaf