intel/zephyr.js #1833: [ashell] 'grovelcd.js' module can't be found when running I2CBMP280.js

xiaoyu3x, 2018-02-07:
## Issue description
'grovelcd.js' module can't be found  when running [I2CBMP280.js](https://github.com/intel/zephyr.js/blob/master/samples/I2CBMP280.js) in ashell. 
In fact, [GroveLCD.js](https://github.com/intel/zephyr.js/blob/master/modules/GroveLCD.js)  has been imported A101.

## Test Code
[I2CBMP280.js](https://github.com/intel/zephyr.js/blob/master/samples/I2CBMP280.js)
[GroveLCD.js](https://github.com/intel/zephyr.js/blob/master/modules/GroveLCD.js)
[BMP280.js](https://github.com/intel/zephyr.js/blob/master/modules/BMP280.js)
## Steps to reproduce
1.Flash ashell into 101 board, open minicom port `minicom -D /dev/ttyACM0`. 
2. Copy [GroveLCD.js](https://github.com/intel/zephyr.js/blob/master/modules/GroveLCD.js) and [BMP280.js](https://github.com/intel/zephyr.js/blob/master/modules/BMP280.js) to ashell.
3. Copy [I2CBMP280.js](https://github.com/intel/zephyr.js/blob/master/samples/I2CBMP280.js) to ashell and run it.
4. When running [I2CBMP280.js](https://github.com/intel/zephyr.js/blob/master/samples/I2CBMP280.js)  on Web IDE, get the same result as ashell

## Actual Result

**Ashell and Web IDE**
(ERROR) failed to evaluate JS, NotSupporttedError: Module not found
[image]

## Expected Result
I2CBMP280.js works normally and no error.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [ede4509](https://github.com/intel/zephyr.js/commit/ede4509d98da5f0187b82a7867fc0277641c4ba7) | Arduino 101 | Feb 7, 2018 | Fail

jimmy-huang, 2018-02-08:
Is this a regression?  Was this broken for 0.5 release or after?

xiaoyu3x, 2018-02-08:
No, it is not a regression.
I just reproduced this issue on 0.5 branch

jimmy-huang, 2018-02-08:
I believe grove_lcd has been turned off by ashell before due to ROM constraints, so it's not on by default, so if you take a look at the components that are enabled by default in src/zjs_ashell.json, you can add grove_lcd to the list and it will fix your problem I think.

xiaoyu3x, 2018-02-13:
This issue isn't reproduced with PR #1846 and it will be closed until PR is merged.

xiaoyu3x, 2018-02-24:
hi,
This issue isn't reproduced when I used ide commit [bae9e36](https://github.com/intel/zephyrjs-ide/commit/bae9e368af04efcafd88bf0185792444ea317ae1) and zephyr.js commit [ed85a4d](https://github.com/intel/zephyr.js/commit/ed85a4deec7fdc4e6c464ba9bef7c03da3339550), so closed it

closed 2018-02-24 by xiaoyu3x