intel/zephyr.js #1842: [k64f] IPv6 address '2001:db8::1' is not assigned

qiaojingx, 2018-02-09:
### Description

This is a regression issue which is caused by zephyr update. IPv6 address '2001:db8::1' is not assigned when running IPv6 Server(UDP, TCP, WS and OCF). These server cannot be connected.
 
### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js)
[UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js) 
[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) 
[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js)
[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js)

### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/TCPEchoServ6.js `
2. `cp outdir/frdm_k64f/zephyr/zephyr.bin /media/<username>/MBED/`
3. add route and address on PC
```
ip -6 route add 2001:db8::/64 dev eno1
ip addr add 2001:db8::2 dev eno1
```
4. `echo "hello" | nc -6 2001:db8::1 4242`

### Actual Result

Connection cannot be established.
 
### Expected Result

Connection can be set up, and server side can receive data "hello".
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4324cf21 - ede4509 | FRDM-K64F | Feb 09, 2018 | Fail
master | ebd7364 | FRDM-K64F | Feb 09, 2018 | Pass
 
### Additional Information

This issue is caused by commit 4324cf2.

jimmy-huang, 2018-03-23:
this might be a zephyr bug, see https://github.com/zephyrproject-rtos/zephyr/issues/6583

grgustaf, 2018-05-17:
I get bisected this problem to ZJS commit 4324cf2, which was a Zephyr update. I further bisected it to Zephyr commit zephyrproject-rtos/zephyr@81ecfc3, which turned off promiscuous mode by default. I also tested that forcing promiscuous mode back on fixes the problem. Now the question is why and whether there's a "right way" to fix this.

pfalcon, 2018-05-17:
Is the description of this issue correct/actual? It says:

> IPv6 address '2001:db8::1' is not assigned ...

So, is confirmed that that this address is not assigned to Zephyr's network interface, or there's only an external symptom that it doesn't respond to netcat? What about ping6s, do they work? @grgustaf said on IRC:

>  i can't seem to ping my k64f after that patch, but if i reenable promiscuous mode it works

It would be nice if the description was updated (or confirmed in comments).

So, all in all, would be nice to get the exact steps to reproduce and the symptoms.

Another thing to keep in mind is https://github.com/zephyrproject-rtos/zephyr/issues/3187 (frdm_k64f may not respond to ping soon after boot).

Otherwise, I'm doing zephyr 1.12 pre-release testing, and hope to try reproducing this.

pfalcon, 2018-05-17:
~~~
***** BOOTING ZEPHYR OS v1.11.0 - BUILD: May 17 2018 20:16:01 *****
Zephyr Shell, Zephyr version: 1.11.0
Type 'help' for a list of available commands
shell> 
Starting TCPEchoServ6.js sample...
listening

shell> select net
net> iface

Interface 0x2000cdc0 (Ethernet)
===============================
Link addr : 00:04:9F:C8:8F:46
MTU       : 1500
IPv6 unicast addresses (max 3):
	fe80::204:9fff:fec8:8f46 autoconf preferred infinite
	2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 2):
	ff02::1
	ff02::1:ffc8:8f46
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 25857
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	192.0.2.1 manual preferred infinite
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 0.0.0.0
IPv4 netmask : 0.0.0.0
net> 
~~~

I.e. the address is assigned actually. But I can confirm that ping6 doesn't work:

~~~
# ping6 2001:db8::1
PING 2001:db8::1(2001:db8::1) 56 data bytes
From 2001:db8::2 icmp_seq=1 Destination unreachable: Address unreachable
From 2001:db8::2 icmp_seq=2 Destination unreachable: Address unreachable
~~~

pfalcon, 2018-05-17:
Actually, not right. Compare with the right picture:

~~~
IPv6 multicast addresses (max 4):
	ff02::1
	ff02::1:ffe6:c52e
	ff02::1:ff00:1
~~~

The fix:

~~~
--- a/src/zjs_net.json
+++ b/src/zjs_net.json
@@ -20,10 +20,11 @@
             "CONFIG_NET_BUF_RX_COUNT=10",
             "CONFIG_NET_BUF_TX_COUNT=6",
             "CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3",
-            "CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=2",
+            "CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4",
~~~

grgustaf, 2018-05-17:
Very interesting... thanks, @pfalcon.

qiaojingx, 2018-05-18:
@pfalcon Update the description. It is speculated according to external symptom that netcat and ping6 both can't get response from address 2001:db8::1.

qiaojingx, 2018-05-18:
Verified with commit 2273038 + pr #1881. TCPEchoServ6.js and UDPEchoServ6.js works fine.
But OcfServer.js, OcfSensorServer.js and OcfMultiServer.js still can't be connect with OCF Client, while ping6 2001:db8::1 is ok.

qiaojingx, 2018-05-24:
Verified with commit f90d9d1. This issue is fixed. For ocf connection can't be established on K64F, a new issue #1885 is created.
Close this one.

closed 2018-05-24 by qiaojingx