intel/zephyr.js #1844: [ashell] Should stm32f4_disco and olimex_stm32_e407 support ashell like ide?

qiaojingx, 2018-02-09:
### Description

Should stm32f4_disco and olimex_stm32_e407 support ashell since they support ide now?
Build ashell target for these two boards failed.
 
### Steps to Reproduction

1.  `make BOARD=stm32f4_disco ashell`
2.  `make BOARD=olimex_stm32_e407 ashell`

### Actual Result

[image]
 
### Expected Result

Build ashell target successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 0ae1d62| stm32f4_disco | Feb 09, 2019 | Fail
master | 0ae1d62| olimex_stm32_e407 | Feb 09, 2019 | Fail
 
### Additional Information

brianjjones, 2018-02-28:
Was just missing a change to the CLI json file, I have a fix open to take care of this.

qiaojingx, 2018-03-09:
Verified with commit 9db8f2b. This issue is fixed.
Close.

closed 2018-03-09 by qiaojingx