intel/zephyr.js #1853: [math] Have a calculation error

cuiyanx, 2018-03-07:
### Description

I test the simple test(following), I find a calculation error, please look at count2~4 and count9 of the logs.
 

### Test Code

Simple test:
```
var valueA = -0.5;
var count = 1;
var valueB = -0.03;

var testTimer = setInterval(function () {
    if (count === 10) {
        clearInterval(testTimer);
    }

    valueB = valueB + valueA * 20 * 20 / 1000;

    console.log("count: " + count);
    console.log("  valueA: " + valueA);
    console.log("  valueB: " + valueB);

    count++;
}, 500);
```

 

### Steps to Reproduction

 

### Actual Result

Logs:
```
count: 1
  valueA: -0.5
  valueB: -0.23
count: 2
  valueA: -0.5
  valueB: -0.43000000000000005
count: 3
  valueA: -0.5
  valueB: -0.6300000000000001
count: 4
  valueA: -0.5
  valueB: -0.8300000000000001
count: 5
  valueA: -0.5
  valueB: -1.03
count: 6
  valueA: -0.5
  valueB: -1.23
count: 7
  valueA: -0.5
  valueB: -1.43
count: 8
  valueA: -0.5
  valueB: -1.63
count: 9
  valueA: -0.5
  valueB: -1.8299999999999998
count: 10
  valueA: -0.5
  valueB: -2.03
```

 

### Expected Result

No calculation error.
 

### Test Builds

| Branch | Commit Id | Target Device | Test Date | Result |
| :---: | :---: | :---: | :---: | :---: |
| master | 85e2d74 | Arduino 101 | Feb 7, 2018 | Fail |

 

### Additional Information

grgustaf, 2018-03-07:
This is not a bug... try running the same thing in Chrome DevTools console (as I just did) and you get the exact same results.

The perceived problem comes from the fact that floating point math has limited precision. So certain values are not representable exactly. The classic example is to hit F12 in your Chrome and go to Console tab (or run `node` on the command line) and then just type 0.1 + 0.2 <Enter>. You'll see 0.30000000000000004.

Here's a good link on the issue: http://floating-point-gui.de/

cuiyanx, 2018-03-08:
@grgustaf OK, I learn it.

closed 2018-03-08 by cuiyanx